opentx/opentx #5853: Fading between Flight phases will not happen

Billy60300, 2018-04-28:
I report that when switching between flight phases no fading will happen. All positions are switched to abruptly.
Taranis QX7, OpenTX 221.

3djc, 2018-04-28:
You need to supply your otx file with instructions for us to reproduce the issue.

Billy60300, 2018-04-29:
> Répondre à :  opentx/opentx
> <reply+02504ebf34f11ada375fecff59398f30cf213e787544705f92cf0000000116fd428b9
> 2a169ce12fe59b9@reply.github.com>
> <author@noreply.github.com>
> (#5853)
>
> QX7 does not use SDcard to save models...
>
> 
> <https://github.com/opentx/opentx/issues/5853#issuecomment-385234736> , or
> mute the thread
> NMSQmT7ks5ttXgLgaJpZM4TrhM-> .
>
>
> b","name":"GitHub"},"entity":{"external_key":"github/opentx/opentx","title":
> "opentx/opentx","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
> 418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
> ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
> -9aed-b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/opentx/opentx"}},"updates":{"snippets":[{"
> icon":"PERSON","message":"@3djc in #5853: QX7 does not use SDcard to save
> models..."}],"action":{"name":"View
> Issue","url":"https://github.com/opentx/opentx/issues/5853#issuecomment-3852
> 34736"}}}

Are you sure ? Mine has a SD-HC card indeed.
I uploaded the OTX file there is on it.
Best regards

kilrah, 2018-04-29:
There is an SD card but it is not used for model storage. Unless you use the "backup" feature in the model selection page which makes a copy on the card, but then those are "archived" and no longer part of the OTX as designed.

This OTX has 2 models. If you see more on your radio's model select screen then please conenct the radio to USB like you would to use companion, but send us the EEPROM.BIN file you will find it the "Taranis" drive that appears on your computer.

kilrah, 2018-04-29:
Your "no fading" is becasue you're using your flight mode switch in the mixes' Switch field. You must leave that blank and use the "Flight Mode" checkboxes.

closed 2018-04-29 by kilrah

Billy60300, 2018-04-29:
Sorry Kilrah, I am afraid I don't get the point.
The Flight Modes depend on SC.
The Mixes which are to happen according to the chosen FMode, should be linked in some way to the FM. If not, how will the program know what mix to activate ? I supposed this link between FM and proper Mix was to be done using the same Switch. Now, I  understand how there is no fading in my case: the SC position will trigger the mixes before the fading has occured.
If I leave the Mix Switches box blank, how will they be triggered ?
Thanks again.

kilrah, 2018-04-29:
I just told you. The Flight modes checkboxes define in what modes the line is enabled.

[image]

Billy60300, 2018-04-29:
Yes indeed ! Got it ! understood !
Everlasting thanks for your help !