opentx/opentx #5854: Models from the SD Card do not appear in OpenTX Companion

Billy60300, 2018-04-28:
OpenTX 221: some models existing on the SDCard do not show up on the Companion after a 'read from the model' call.

kilrah, 2018-04-28:
What radio? What models?

kilrah, 2018-04-29:
There is an SD card but it is not used for model storage. Unless you use the "backup" feature in the model selection page which makes a copy on the card, but then those are "archived" and no longer part of the OTX as designed.

This OTX has 2 models. If you see more on your radio's model select screen then please conenct the radio to USB like you would to use companion, but send us the EEPROM.BIN file you will find it the "Taranis" drive that appears on your computer.

Billy60300, 2018-04-29:
Thank you for the explanation. Here is the EEprom file.

[eeprom-x7.zip](https://github.com/opentx/opentx/files/1958576/eeprom-x7.zip)

kilrah, 2018-04-29:
This comes from the simulator and not from your radio and only contains one blank model. You need to take the one from the radio's USB drive that is named exactly as I mentioned.

Billy60300, 2018-04-29:
Here it is.
Thanks again
On this one in Companion, I see all my models fine.
[eeprom-2018-04-29-092513.zip](https://github.com/opentx/opentx/files/1958620/eeprom-2018-04-29-092513.zip)

kilrah, 2018-04-29:
This is again not the requested file. How did you get that one....?

Billy60, 2018-04-29:
Well, I looked in the Taranis directory for the latest EEprom file after saving the SD card from the TX.
I worked for a long time from the EEprom content on the Companion and simulator and could implement all the new recommendations you gave me and it worked beautifully on the simulator.
I tried to upload the files to the TX and some of the changes I had made did appear, but not all ... ?
So I decided to reprogam directly on the TX. Now it works fine on the TX.
I wish the relationship between the SD card, the 'Save' command, the EEprom Files, the archiving system and so forth were explained somewhere. I would have probably avoided lots of mistakes !
Thanks for your help !