opentx/opentx #5861: V2.2.2Nxxx Simulator at companion ofte starts not correct, V2.2.1 was ok

hrenz, 2018-05-01:
look at video (zip-file)
simulator start, switches allways work ok
channels often not working,
have to restart often, sometimes it works after 2 or 3 restart
same at X9, X10,X12

[Video_2018-05-01_085933.zip](https://github.com/opentx/opentx/files/1963174/Video_2018-05-01_085933.zip)

mpaperno, 2018-05-16:
This might be a dupe but it's the first post which clearly shows what is going on.  For example until I watched the video I had no idea "reload" meant the reload button in simu UI (vs. restarting the whole simulator app).  Definitely a clue there.  Are we sure this is even the same issue as 4520...? Seemed like some red herrings in that one and no conclusive repro case anyway.

hrenz, 2018-06-19:
ok, next video.
after start, Simulator worsks all time well, 
close Simulator, change something, open simualtor again
you see: switches works allways well, all other things works too,
only the analog part works most time not,  
or at second or third time simulator  open  the analog part dont work  
[Video_2018-06-19_190458.zip](https://github.com/opentx/opentx/files/2116194/Video_2018-06-19_190458.zip)

hrenz, 2018-06-19:
its video is from today with opernTx V2.2.2 RC1 
but this issues is at V2.2.1 and  at V2.2.0 too

hrenz, 2018-06-19:
reopen

kilrah, 2018-06-19:
Post your otx file

mpaperno, 2018-06-19:
Thanks for keeping with this.  I don't know why but this seems very hard to recreate here (I have never been able to).  These kinds of issues are the worst.

In addition to the OTX file, I'd also like to see the CPN/SIM debug logs, if you could.  You can enable those in _Companion -> Edit Settings -> Application Settings -> Debug Output Logging_ (at the bottom).  

Please enable both options (Application and Radio Firmware).  Then restart Companion, and just do the minimum necessary steps to reproduce the problem (this way the logs won't get too huge, hopefully). 

Thanks again.

mpaperno, 2018-06-19:
Does this ever happen when simulating the whole radio, or only with individual model(s)?

Also in the latest video you made a change in the model before launching the 2nd time -- is this "required" for the problem to manifest?  In other words, if you don't make any changes before launching the 2nd time, does the problem still happen?

hrenz, 2018-06-30:
its indenpendent of changing or not something in inputs or mixers.

its "only" the analog part that ist somtimes not starting, or not setting/resetting correct
switches, log switche, timers allways works

mpaperno, 2018-07-01:
Still need more information please, as requested.  

* Does this ever happen when simulating the whole radio, or only with individual model(s)?
* OTX file with at least one model that has this issue.
* Windows version.
* Debug logs.

Thanks.

hrenz, 2018-07-01:
ok, here its a much longer video from today,  5 min
with lots of open and close the simulator,
sometimes it works, sometime not,  
switches works allways, log switches and analogs not allway
I don´t change things at programm, 
only open, show simulator, close, open simu, 
with debug window

Its independemt from Modell, Programm or  Tx  I select

Win 7 Prof
CompanionV2.2.2RC2
[Video_2018-07-01_114048.zip](https://github.com/opentx/opentx/files/2152703/Video_2018-07-01_114048.zip)

kilrah, 2018-07-01:
You are still not saying if you also have the issue when using "simulate radio" in the document window instead of the "simulate" button in the model edit window.

hrenz, 2018-07-01:
ah, ok , never tested simulate radio, will do now