opentx/opentx #5865: X7 boots from bootloader only (2.2.1)

diggit, 2018-05-02:
Hi, I have encountered some weird behavior of my Taranis X7.

With opentx 2.2.0 X7 boots up normally.
With 2.2.1, only green LED flashes after pressing power button and it stays off. However, it is possible to boot in bootloader and after selecting exit, my X7 boots into opentx.
Same thing happens when I build it myself. My options are: `cmake -DPCB=X7 -DGVARS=YES -DLUA=YES -DHELI=OFF -DBINDING_OPTIONS=ON -DSUPPORT_D16_EU_ONLY=ON -DCMAKE_BUILD_TYPE=Release -DDISABLE_COMPANION=YES`.

The I tried bisecting this and commit introducing this behavior is: 4c40aec8b38f043fb7bfa8a2a665f724c0488115.

Did anybody else encountered such problem?

I know at least about one other person ([YT](https://www.youtube.com/watch?v=vnAwAMTl7U0) in comments)
> **RC and FPV Jonathan**
i have the same after installing opentx 2.2.1 it will start up if you put in in bootloader mode and then exit. not sure why tho..﻿

3djc, 2018-05-02:
Keep pushing power button. It starts like an X9E now and power neeeds to be maintained

diggit, 2018-05-02:
Oh. Thanks :)

closed 2018-05-02 by diggit