opentx/opentx #5871: ERROR FLASH OPENTX - TURNIGY 9XR PRO 2018

raf77raf, 2018-05-04:
The TURNIGY 9XR PRO 2018 Flash with OpenTX, but I'm having a problem.
The problem occurs for the new tested versions of OpenTX: "2.0.20, 2.1.9, 2.2.1".
SD card flash it displays a message: "EEPROM warning - wrong EEPROM data - Press any key".
After the EEPROM is installed, the EEPROM Recording process will appear, by means of a blinking screen after a while. in this context, without starting OpenTX.

Ma makes a Flash with the ER9X version 221, loads normally.

You can see the problem occurring on YouTube Video: https://youtu.be/mAB-E4ekh_g

Could someone help me with this problem?

#################### TEXTO EM PT-BR ####################
Estou tentando atualizar a Flash do TURNIGY 9XR PRO de 2018 com o OpenTX, mas estou tendo problema.
O problema ocorre para as seguintes versões testadas de OpenTX: "2.0.20, 2.1.9, 2.2.1".
Ao concluir o Flash pelo cartão SD ele aparece a mensagem de: "EEPROM Warning - Bad EEPROM Data - Press any key".
E após confirmar a instalação da EEPROM, parece que começa o processo de Gravação da EEPROM, porem a tela da um piscada depois de um tempo. fica neste Loop, sem iniciar o OpenTX.

Ma quando faço a Flash com o ER9X versão 221, carrega normalmente.

Pode verificar o problema ocorrendo no Vídeo YouTube: https://youtu.be/mAB-E4ekh_g

Alguém poderia me ajudar com este problema?

rubenix, 2018-05-04:
@raf77raf I own same radio (9xr pro) and using opentx 2.2.1 with no issues so far.
I would flash the firmware using companion TX --> It will download and write the right firm to radio :+1: 
Downloadable from: http://www.open-tx.org/2017/12/16/opentx-2.2.1
Don't forget to download and write to SD: **SDCard content for 2.2.1 (2.2v0013)**

closed 2018-05-07 by raphaelcoeffic

rubenix, 2018-05-09:
I could only think of bad sd card¿?

raf77raf, 2018-05-09:
I tried both ways, with the Companion and SD.
Both ways gave the same error.

raphaelcoeffic, 2018-05-09:
@raf77raf the msg you see `EEPROM warning - wrong EEPROM data - Press any key` is absolutely normal the first time you install OpenTX, as the EEPROM must be formatted. However, what happens after that (blocking at start) is not normal. However, none of the devs does have such a TX, which basically means that we can offer no support. ErSky9X is very well supported on that TX, so why not using it?

raf77raf, 2018-05-09:
In that so much use of OpenTx or ErSky9X, is that I have been using OpenTx for a long time, so I have a lot of familiarity like OpenTx, something I'm having to learn as ErSky9X.
I'm using ErSky9X already about 20 days, I can say that I still prefer OpenTX, because of the features in the functions.