opentx/opentx #5878: Wrong current and consumption reading from FrSky ESC-80A sensors

RiMr71, 2018-05-08:
Tested ESC80A in FrSky Vantac AR+ on OpenTx 2.2.2N597.
Wrong data from current sensor, wrong data from consumption sensor - compared with standard 40A sensor. 
Missing data in log file.

3djc, 2018-05-08:
Given the amount of details submitted, we have very little chances to understand, let alone fix the issue !

RiMr71, 2018-05-08:
FrSky speed controller with in-build sensors FrESC32_80A. Horus X12s with OpenTx 2.2.2N597. In telemetry windows are detected right sensors from this ESC (rpm, current, power consumption, temperature), but values from sensors current and power consumption are not valid (numbers are with values which are not true). If is it still hard to understand (sorry for that), I will make a video.

3djc, 2018-05-08:
Any details on the value measured vs expected ?

kilrah, 2018-05-08:
As long as you have your offset and Range set to 0 what OpenTX displays is what the sensor sends. If the value is wrong then the sensor sends a wrong value. Not much we can do.

Please send your otx file as well.

3djc, 2018-06-20:
I am leaving this open until final confirmation by FrSky, but it looks like the issue is not with OpenTx

notlebowski, 2018-07-10:
After insisting there was a problem with the ESC it was confirmed by Frsky some time ago.
https://www.rcgroups.com/forums/showpost.php?p=39849336&postcount=80

Allerhopp tested this new firmware and saw that although values are now decent, they still are not correct.
https://www.rcgroups.com/forums/showpost.php?p=39894448&postcount=88

Frsky will offer a new firmware soon.
https://www.rcgroups.com/forums/showpost.php?p=39898924&postcount=97

Problem is obviously not in OpenTX
I think the ticket can be closed.

3djc, 2018-07-11:
thx for the update 👍

closed 2018-07-11 by 3djc