opentx/opentx #5888: Companion 2.2.x AVR Boards selectable for FW Download

r41065, 2018-05-14:
http://www.open-tx.org/2018/02/26/avr-pull
There is written that AVR Boards are not longer supported by OpenTx newer 2.1.x.
This boards are still selectable in Companion with their Options for Building Firmwares .
But Trying to Download Firmware always result in Error " invalid Board "

will there an Solution to get AVR Boards supported again ? If not please delete these Boards from the Radio Selection in higher Companion Versions....

Ralf

3djc, 2018-05-14:
There is currently no plan to add support back for those, but they need to stay in the menus because you can import the settings for newer radios

closed 2018-05-14 by 3djc

r41065, 2018-05-14:
Thanks for Information

Ralf