opentx/opentx #5889: Polish characters in Horus - OpenTX 2.2.1

oizic, 2018-05-14:
I have uploaded OpenTX 2.2.1 to my Horus
There is a problem with Polish characters.
Instead of Polish characters, there are various inappropriate symbols.
[image]

oizic, 2018-05-14:
That's how Polish signs are missing
There are no Polish characters on this list.
When will they be introduced?

raphaelcoeffic, 2018-05-15:
Probably when some polish fellow starts working on it? Or at least helping? That‘s how open source works ;-)

3djc, 2018-05-15:
We probably need to review the fonts and display routine to deal with polish display ourselves, but yes; polish translations have not been reviewed in a loooonng time, so likely need quite a bit of work !

oizic, 2018-05-15:
I do not know how to program, so I temporarily use the English menu.
Only I wanted to report a problem in the Polish menu.

commit c6f58543 (2018-05-16): [horus]: fixed international characters for es, fi, it, pl, pt & se

fixes #5889