opentx/opentx #5899: OTX 2.2.1, Taranis X9D - SF:Adjust GVx:Source: TrmX problem

spoke2570, 2018-05-21:
Taranis X9D OTX 2.2.1 While using Global variable set to precision 0.0, range -1024% to 1024%
Special Function:Adjust GVx - Parameters: Source, TrmX 

GVx adjustable range is -12.5% to 12.5% 
Global variable precision 0._ gives +/- 125%
It seems limited by maximum trim tics not actual value.

Would be nice to open this as a nice increment/decrement tool that has access to the new better resolution, without having to use 2 special functions per GVx to modify the value. 

I'm calling it a problem 'cause I think you should be able to see the same value range despite the accuracy change.

Thanks for all your hard work!

spoke2570, 2018-05-21:
..and if one is to use the increment +/- 0.1 then a re-press of the trim tab is needed for each tenth

grumble

kelkarakshay, 2018-05-26:
Hi friends how mtwomey add cross trim in global function. Is it any other versions of opentx. I have installed latest opene tx 2.2.1. And I m unable to find it. Let me know if there is another versions. I want to install it.

kilrah, 2018-06-16:
When you use Source/TrmX you are setting the GVAR to the value of the trim register.

closed 2018-06-16 by kilrah