opentx/opentx #5903: X9D+ with 2.2, .2.2.1. The master/Jack doesn't work but slave/Jack works.

wuhao19920619, 2018-05-21:


raphaelcoeffic, 2018-05-21:
@wuhao19920619 On your video, it is not clear which type of a signal you have on the other end of the cable. If you don't have any, then it's clear it's not going to work. The best place to have a look is the trainer page in the radio settings menu. Please note that this is not a support forum, but an issue tracking system. Please use rcgroups or some other forum of you liking next time.

closed 2018-05-21 by raphaelcoeffic

wuhao19920619, 2018-05-21:
@raphaelcoeffic I think this might be a bug because at least there should be an icon for indication. I tried a headtracker with CPPM it works fine but with jack nothing happens. So I believe there must be something wrong...

raphaelcoeffic, 2018-05-21:
@wuhao19920619 many people had issues until they used the right cable (don't remember, might have to be mono cable).

kilrah, 2018-05-21:
If no icon shows then no proper signal is received, so check the cable and whatever device you tried to connect but didn't say what it was.

wuhao19920619, 2018-05-22:
@raphaelcoeffic Yes you are right, After I use a mono cable everything works good. And I see that with slave mode once the cable insert then there will be aan icon. But with master mode the icons will goes out only if there is a ppm signal going in. Thank you guys @kilrah @raphaelcoeffic