opentx/opentx #5905: Function request: Adjust GV in global function is missing on the remoter.

wuhao19920619, 2018-05-21:
The Adjust GV function works fine in companion 2.2.0 but missing in  companion 2.2.1. And on the remoter no matter with which firmware there is no "adjust GV" in global function. But if you set it in companion 2.2.0 then it works good in the remoter.

I think this is pretty useful especially to define a switcher as a maximum range selector for all models.

raphaelcoeffic, 2018-05-21:
Do you mean that one? this is on Companion 2.2.1:

[image]

wuhao19920619, 2018-05-21:
yes, what I meant is in the global functions. So it works for all models.

wuhao19920619, 2018-05-21:
Yeah I guess so but I don't know the reason. 
One interesting thing is that in 2.2.0 this is only adjustable in companion. In the remotor you cannot find it either......

wuhao19920619, 2018-05-21:
global variable is "global" functional......So I think logically it should be adjusted also globally.

raphaelcoeffic, 2018-05-21:
@wuhao19920619 that's basically the issue, GV are anything but global ;-) They have a different value per flight mode, and thus per model. As it seems, it was never intended to be possible to edit GVs from these gobal functions: https://github.com/opentx/opentx/commit/1c1624b8a6e220896401442e750940d230dcde2b

closed 2018-05-21 by raphaelcoeffic