opentx/opentx #5907: French Name # English Name for Sensor

aviat40, 2018-05-24:
Hello,
I am developping a Widget and noticed that the name of the sensor for the Battery RX is not the same according to the language of the radio.

It is "BtRx" in French and "RxBt" in English.

I think all sensors should have the same name in all languages because when you want to use it to display on a widget, you need to write a special function to detect which language is used.

Thank you
Thierry

raphaelcoeffic, 2018-05-24:
@aviat40 Whereby I think you're right with the default names being the same with all languages, you should consider make the sensor name a config parameter of your widget, so that people can use renamed / other sensors as well.

3djc, 2018-05-24:
As I replied in the French forum, you need to allow user selection, as sensor are designed to be renamed.

If Lapinfou wants to change the French translation, why not, but that won’t change the fact that you shouldn’t assume names

lshems, 2018-05-24:
Could we increase the amount of parameters to pass through to a widget.

It's becoming troublesome if we have more then one sensor to define e in a widget setup.

LapinFou, 2018-05-24:
@raphaelcoeffic I may change the translation but, as mentioned @3djc , it would a better (and more reliable) solution to update the widget.

aviat40, 2018-05-24:
Ok, May be change the translation initially to have the same name sensors and I will also update the widget as 3djc advises ;-)

3djc, 2018-05-24:
If that french name has been used for a long time, then I would advise against changing it (since I don't ise the french version, I can't tell)

kilrah, 2018-05-24:
I am against changing the translations.
Even if we did it would not solve your problem, you'd still need to handle French names becasue the change would only affect new users/newly configured models

aviat40, 2018-05-24:
@kilrah : ok, I already start to manage with the French. It seems that was not a good idea. So please for LapinFou how translate, change also within companion simulator sensors. 
Any case, thank you for all of your answer

LapinFou, 2018-05-25:
@aviat40 
Je vais écrire en Français afin d'éviter toute ambiguïté. Comme @3djc l'a dit je suis **contre** l'idée de changer la traduction. Toutes les versions OTX supérieure à la 2.1 permette de nommer/renommer les capteurs comme on veut. Donc, c'est au widget de faire en conséquence.
Maintenant si cela te bloque complètement dans le développement de ton widget et que tu n'as pas d'autre solution, alors dans ce cas je met à jour la traduction. Cependant, je ne gère que la traduction Française de **Companion** !! Pour le firmware radio, il faut convaincre @3djc/@kilrah/@bsongis ...
😉

3djc, 2018-05-25:
The issue was raised rightfully as he saw that as an issue. It as become clear that it is not quite the case, I think he understands that clearly and will adapt his widget as a result of those discussion, therefore, closing this. thx to all :)

closed 2018-05-25 by 3djc