opentx/opentx #5908: X-Lite SD card contents

teckel12, 2018-05-25:
Trying to test my INAV telemetry script on the X-Lite but can't find the SD card contents for the X-Lite transmitter.  As 2.2.1 support doesn't seem to be included, I'm running the nightly build, but https://downloads.open-tx.org/2.2/nightlies/sdcard/ doesn't include the X-Lite.

Could someone point me in the right direction?  I tried searching this repo, open-tx.org site, Google search, etc.  The configurator points me to: https://downloads.open-tx.org/2.2/rc/sdcard/opentx-xlite/ which is empty.  When I try to download the firmware from the configurator I get a "Compilation error".

Thanks!

teckel12, 2018-05-26:
It seems there isn't special SD card contents for the X-Lite?  Not sure why the firmware won't download.

closed 2018-05-26 by teckel12

3djc, 2018-05-26:
currently, you should use : https://downloads.open-tx.org/2.2/nightlies/sdcard/sdcard-taranis-x7-2.2V0016.zip

teckel12, 2018-05-27:
@kilrah Bingo! That's what I forgot I believe, thanks!