opentx/opentx #5911: LUA lcd.drawPixmap() on X7(s)    y coordinate don't work

mordicus44, 2018-05-26:
ex : lcd.drawPixmap(4,Y,imgName1) 

only valid values for Y coordinate are : (1,8,16,24,32,40,48,56,64)
other values have no effect on image position

Best regards Morgan

MorganTr44, 2019-08-17:
No issue to draw img at à wanted positions ?

raphaelcoeffic, 2019-08-27:
@MorganTr44 the issue here is that the framebuffer uses 1 bit per pixel, which means that it would require quite some shifting to write at a specific position. The draw routine is made to write into the frame buffer aligned on bytes, which is faster.

So, whereby it would be possible to "fix", this seems to me to be "by design", and as it is used everywhere, it would possibly move quite some stuff around on existing screens, which means re-testing and adjusting everything on these 1bit displays.

@bsongis should we really change that? Or offer a second function which is pixel exact just for LUA?

raphaelcoeffic, 2019-08-27:
here is the function for reference:
```c++
void lcdDraw1bitBitmap(coord_t x, coord_t y, const uint8_t * img, uint8_t idx, LcdFlags att)
{
  const uint8_t * q = img;
  uint8_t w = *q++;
  uint8_t hb = ((*q++) + 7) / 8;
  bool inv = (att & INVERS) ? true : (att & BLINK ? BLINK_ON_PHASE : false);
  q += idx*w*hb;
  for (uint8_t yb = 0; yb < hb; yb++) {
    uint8_t *p = &displayBuf[(y / 8 + yb) * LCD_W + x];
    for (coord_t i=0; i<w; i++){
      uint8_t b = *q++;
      if (p < DISPLAY_END) {
        *p++ = inv ? ~b : b;
      }
    }
  }
}
```

bsongis, 2019-08-27:
I think we should keep the optimization when we are aligned, but in all other cases we should do the right job

raphaelcoeffic, 2019-08-27:
Please note that this might change the position on the screen of quite some items in the 128x64 res.

bsongis, 2019-08-27:
If this is the case, we could add another function, just used by Lua (which would call the current function in case alignment is good)

raphaelcoeffic, 2019-08-27:
yes, I'd really prefer that!

raphaelcoeffic, 2019-09-02:
@MorganTr44 Could you please upload such a pixmap? at the moment, I get only garbage with the current code and a 1bit bitmap (24x40) produced with Gimp.

PR #6703 (2019-09-03): [X7] fix #5911 and bmp loading
Tested with this script:
```lua
y=12

local function runFunc(event)

  lcd.clear()
  lcd.drawText(1,1,"drawPixmap() example", 0)
  lcd.drawPixmap(24, y, "/delta.bmp")

  if event == EVT_VIRTUAL_NEXT then
     y = y + 1
  elseif event == EVT_VIRTUAL_PREVIOUS then
     y = y - 1
  end
  return 0
end

return { run=runFunc }
```

and this bitmap (github does not take bmp files...):
[delta.bmp.zip](https://github.com/opentx/opentx/files/3567183/delta.bmp.zip)

closes #5911
commit 27946ee9 (2019-09-03): [X7] fix #5911 and bmp loading (#6703)

* lcdDraw1bitBitmap now bit exact

* bmp rows are padded up to a multiple of 4 bytes
closed 2019-09-03 by bsongis