opentx/opentx #5912: On X-Lite Lua lcd.drawFilledRectangle FORCE and ERASE don't appear to work

teckel12, 2018-05-26:
Using the latest nightly build the `lcd.drawFilledRectangle` FORCE and ERASE arguments don't appear to work on the X-Lite (at least in the OpenTX Companion).

A Lua script that renders correctly on the Q7X and X9D renders incorrectly on the X-Lite.  I can attach pictures later if there's an issue identifying the problem.

raphaelcoeffic, 2018-05-27:
I believe this should be fixed with https://github.com/opentx/opentx/commit/535e41d43e54248629af6f4f79307e81bd7075f3. Please try the latest nightly.

kilrah, 2018-05-27:
Duplicate of #5909.

closed 2018-05-27 by kilrah

teckel12, 2018-05-27:
This could be different as with the drawLine command it threw an error about argument 6, with drawFilledRectangle it doesn't throw an error but then again just doesn't respect the option. Also, that fix seems to duplicate code already in the file, so I'm not sure about it.

I'll test the latest nightly and if I have a. Issue maybe I'll try again to create a build environment for OpenTX (I wasn't successful previously).

kilrah, 2018-05-27:
Similar code may have already been in the file, but applied to other radio types and wasn't included for X-Lite..

teckel12, 2018-05-27:
@kilrah Tested with the latest nightly and all is good.  The only modifications that are needed for Lua  scripts is likely the new event key names for the X-Lite.  For example, ENV_UP_BREAK instead of ENV_PLUS_BREAK.

In any case, all is good with a few simple additions to support the X-Lite.  Thanks!