opentx/opentx #5916: X-Lite Safety Power-ON Function

dayue1984, 2018-05-29:
To prevent the x-lite from accidentally opening in bag, could you add a function like this:
1.Short push the power button to active it.
2.Then release the button in 1sec, otherwise system will stop power-on procedure.
2.After you release the button, long push it again to finally power on.

Thanks!

lshems, 2018-05-29:
Please keep things simple. Make your own physical solution if you need that. A switch cover or something like that.

We've had enough trouble with x7 power on and off delays.

In competition, a spare transmitter with craft should be powered on as quickly as possible. Every second does count.

3djc, 2018-05-29:
@dayue1984 We have added long push to start x7 and xlite for that very purpose, in a search for the right balance between too much and too little. If this is not enough for you (nothing wrong with that) , then maybe you can easaly remove one of the battery, ensuring therefore radio wont turn on

dayue1984, 2018-05-29:
Clear!Thank you all for the great works!
It's just my personal advise, maybe a little too complicated.
Case closed. Thanks!

closed 2018-05-29 by 3djc

kilrah, 2018-05-29:
It's the "DJI-style" power sequence. Fine, but since we already have people complaining that the current method is too long and complicated we won't go there.