opentx/opentx #5918: Zadig Tip of the day (Remove?)

RCdiy, 2018-05-29:
[image]
Without connecting the radio 
 This simulates faulty USB cable, damages USB port, a number for USB problems.
Write firmware to radio after download check box selected

Click on File > Download
Save
Then write firmware
After progress bar window
Window displayed
 **Tip of the day
 Your radio does not seem to be connected to USB or the driver is not installed!!!
 Please use ZADIG to properly install the driver.**

The sentence about Zadig leads may people down the wrong path. These days for Taranis Zadig is no longer required depending on boot loader version.

Wheather or not to install Zadig requires more investigation.

**Suggestion 1: Remove the ZADIG sentence.**

Suggestion 2: Replace the ZADIG sentance with "Please post in the forums for help."

schwabe, 2018-05-29:
When flahsing this way you need the zadiag driver

RCdiy, 2018-05-29:
No. You may be missing the point. A faulty usb cable will give this error. For example there are cables used for charging with either the data missing or data wires shorted. 

So when you say flashing this way what do you mean? In my original post there is no flashing. The connection was failed to be made. 

I said without connecting the radio. 

These days more information or investigations are needed before installing zadig.

kilrah, 2018-05-29:
His point is that companion falls back to trying dfu if mass storage isn't found, and pops up the message if still not found. That can mislead people into a nightmare of having to try to load the driver when they should rather just solve the problem that casues mass storage not to be found.

commit 6d649279 (2018-05-30): Remove potentially misleading message. Fixes #5918
closed 2018-05-30 by 3djc