opentx/opentx #5923: Unable to Simulate Telemetry for Cels/LowC/CelD

0zeroth, 2018-05-31:
Using OpenTx 2.2.2, simulating Taranis QX7. On the physical radio, it is set up and correctly receives battery telemetry, using the Cels/LowC/CelD sensors. These are configured like [Name: Cels, Type: Custom, ID: 0300 2, Precision: 0.00, Logs: No]. When reading this model into OpenTx companion, this configuration is copied across successfully.

When simulating the radio, and simulating the telemetry, the values for Cels/LowC/CelD to not appear to simulate at all. The RxBt and RSSI do simulate. The screenshot show the behaviour (I have also tried varying all the cell volatages with no success).

I am only trying to type the values, not replay a log file. The exact use case is testing a low volatage alarm based on the LowC value (ie lowest cell volatage). This works fine on the physical radio, but I can't reproduce in the simulated radio.

[image]

3djc, 2018-05-31:
There is no 2.2.2 yet, so it is unclear what version you are using

0zeroth, 2018-05-31:
Typo, it's 2.2.1

Anyway - I started from scratch reloading the models from the radio, and this time it worked. Unsure what I got wrong first time around, but this seems most likely user error. If I manage to reproduce I'll reopen

closed 2018-05-31 by 0zeroth

kilrah, 2018-05-31:
The firmware filters values because of a faulty FLVSS firmware back in the day, so if you type in unrealistic values that don't match a lipo cell (don't remember the criteria anymore) the firmware will ignore them.