opentx/opentx #5927: OpenTX 2.2.0 Timer controlled by logical variable L01-L64 has a strange behavior

WillyZehnder, 2018-05-31:
The controlling of a Timer (I just checked Timer 1) by a Logical Variable has a different behavior depending on the Logical Variable you use:
L01 - L37 : correct function
L38 - L39 : no function
L40 - L46 : inverted function!
L47 - L64 : no function

[image]

3djc, 2018-05-31:
Please test with latest version in this branch, 2.2.1 for now

WillyZehnder, 2018-05-31:
Yes, I thought about that, but I decided not to change my running system because of the challenging situation in private and on the job in the moment. I'm sorry.
If somebody has 2.2.1 on his system it would be nice to make that little test for us.

kilrah, 2018-06-01:
You also don't say what radio you have.

WillyZehnder, 2018-06-01:
Oops, sorry - I use an X9E
But in the meantime I recognized that you can simulate that effect in Companion, too.

kilrah, 2018-06-01:
In what companion version?...

WillyZehnder, 2018-06-02:
It's Companion Version 2.2.0 May 30 2017 (that is fitting to my OTX Version 2.2.0).
Thank you for you patience - I should have known all the questions before you asked me. I developed software for 16 years but I'm a newbie in that GitHub-forum.

commit a0c2f0bf (2018-06-16): Timer mode overflow (fix #5927)
closed 2018-06-16 by kilrah