opentx/opentx #5929: Simulator tweet

nickbraithwaite, 2018-06-01:
When I open the simulator on the Telemetry page I hear a tweeting, 3 sounds at one pitch followed by three at a slightly higher pitch.  This is repeated every few seconds.  I can't find out what this is nor how to stop it. I am running Version 2.2.1, Dec 17 2017

RCdiy, 2018-06-01:
Have you posted in a forum to try and debug?
Issues are for bugs. Not help. 
When posting an issue it helps the developers if you can explain how they can reproduce the bug. For example attaching a zip of the otx file. 
If you do post in a forum please close this issue.

kilrah, 2018-06-01:
Low battery warning probably.

closed 2018-06-01 by kilrah

nickbraithwaite, 2018-06-02:
Many thanks.  Will do.

Nicholas Braithwaite
nicholasbraithwaite@me.com
Box 401, Hahndorf
S. Australia 5245

nicholasbraithwaite.com.au

https://www.amazon.co.uk/So-what-does-conductor-music/dp/064697145X/ref=sr_1_1?s=books&ie=UTF8&qid=1515381388&sr=1-1&keywords=nicholas+braithwaite

nickbraithwaite, 2018-06-02:
Thanks for this - :  I am out of my depth here, but there is no battery influencing OpenTx Companion on my Mac as far as I am aware.  Simulator seems to work, but the noise is a) irritating and unexplained and b) a worry.

ATB

Nick

Nicholas Braithwaite
nicholasbraithwaite@me.com
Box 401, Hahndorf
S. Australia 5245

nicholasbraithwaite.com.au

https://www.amazon.co.uk/So-what-does-conductor-music/dp/064697145X/ref=sr_1_1?s=books&ie=UTF8&qid=1515381388&sr=1-1&keywords=nicholas+braithwaite

mpaperno, 2018-06-02:
The battery voltage is simulated to be a "typical" valid voltage for the radio type being used. This is what is shown in the voltage display. If it is below the alarm threshold set up in the radio settings (of the radio file being used) then the alarm will go off.  So the options are to adjust the radio alarm settings, disable the alarm, or turn down the volume. :)

nickbraithwaite, 2018-06-02:
Thanks!  All clear.  Is there no way to persuade Companion that the battery has more than 7.0 volts?  I don’t want to lower the radio alarm setting - or disable it - for the actual radio, and it would be convenient if I don’t have to turn down the volume - I like to hear all the other audio responses.

Your help much appreciated!

Nick

Nicholas Braithwaite
nicholasbraithwaite@me.com
Box 401, Hahndorf
S. Australia 5245

nicholasbraithwaite.com.au

https://www.amazon.co.uk/So-what-does-conductor-music/dp/064697145X/ref=sr_1_1?s=books&ie=UTF8&qid=1515381388&sr=1-1&keywords=nicholas+braithwaite

mpaperno, 2018-06-02:
> When I open the simulator on the Telemetry page

Wait, this ONLY happens when you're using the Telemetry Simulator (with the Simulate checkbox enabled)?  Do you have some other alarm set up?

Or do you mean some other Telemetry page (on the simulated radio UI itself)?

kilrah, 2018-06-02:
> Actually I just checked the code again and it should be automatically adjusting the simulated voltage based on the radio battery min/max range settings. 

It is now for 2.2.2, but there was a bug with that before.

nickbraithwaite, 2018-06-03:
Aha!  I have 2.2.1.  I find I can deal with it temporarily by altering the Battery Calibration to something within the parameters of the Battery Meter Range.  It reverts if I close the file.  Not a big issue, so perhaps I should wait until 2,2.2. is accessed by “Check for updates”?

Very many thanks for your help

Nicholas Braithwaite

nickbraithwaite, 2018-06-03:
> Begin forwarded message:

Reply to your :
The battery voltage is simulated to be a "typical" valid voltage for the radio type being used. This is what is shown in the voltage display. If it is below the alarm threshold set up in the radio settings (of the radio file being used) then the alarm will go off. So the options are to adjust the radio alarm settings, disable the alarm, or turn down the volume. :)

Many apologies for “spamming”  Won't happen again.

Andre Bernet sorted it for me -see below