opentx/opentx #5932: SF Play Track not played after flight reset although condition becomes true

notlebowski, 2018-06-01:
Applies to:
X-Lite OpenTX 2.2.2N619
X9D+ OpenTX 2.2.2N619
X9D+ OpenTX 2.2.1

I have created a model for my X-Lite with a throttle safety.
To keep the setup organized I used flightmodes.
FM1 has the greatest priority and in that flightmode I have a "last replace line" in the CH3 with weight -100

FM1 is based on !L01 which is a sticky.
Because a sticky is FALSE unless triggered this is my way for making sure that the model directly enters in FM1.

I want to have a vocal warning that FM1 is active/inactive.
For this I created 2 SF's

SF1  !FM1 Play Track "thract" 1x
SF2   FM1 Play Track "thrdis" 1x

**The whole system works as I want it, but during testing/playing of the setup I noticed that sometimes it doesn't say "Throttle Active"**. At first it seemed random. It took some time before I found when I could let it consistently fail one time. It was after a "flight reset".

In the display I can see the flightmode it is in and it starts the model in "Safety" just like it does after the initial model selection, but when I enable FM0 by pressing "elevator trim down" I can see the flightmode change to "Normal" (FM0), but the track is NOT played.
Subsequent enabling of FM0 are correctly called out.

Deliberately I have configured the SFs to get triggered on a logical condition detecting the action instead of triggering them on logical condition that triggered the action.
In this case it means I have SF1 triggered by !FM1 and not triggered by L01

If I have SF1 triggered on L01 it DOES play after a "reset flight".
I think it's because of the "NOT"

- Can you confirm/deny if it is doing this on your devices?
- If so, do you have an explanation for it?
- Do you agree it should not be doing this?

[Kinetic-01-06-2018.otx.txt](https://github.com/opentx/opentx/files/2063807/Kinetic-01-06-2018.otx.txt)

notlebowski, 2018-06-13:
There's currently an RC.
Before going to a release could someone check my model file to verify why SF1 is not being executed after a "Flight reset"? I noticed it during Companion sessions. Disconnecting the USB-lead is the same/similar to a flight reset.

This means that the state in which the radio is in after a "Flight reset" is not the same as the state it's in after selecting a model.
IMHO this should be worrying.

I'm still open to any explanation to why it doesn't behave as I expect it to, but until now no comment is given. This setup is evolving and besides this little thing it does all I expect it to.

https://www.justfly.solutions/index.php/full-repository?view=product&id=20:kinetic

kilrah, 2018-06-13:
> SF1 !FM1 Play Track "thract" 1x
SF2 FM1 Play Track "thrdis" 1x

This is incorrect, you didn't set "1x", you set "!1x" aka "play once, not during startup". So it's normal it's not played during flight reset since that's like a "startup".

notlebowski, 2018-06-14:
Please re-open.
The !1x or 1x doesn't make a difference. The behaviour is the same.

But even if then it should still say "throttle active" when going from FM1 to FM0.
The radio is already started at that moment.

It doesn't behave this way after the model is normally selected.
It only does so after a reset.

notlebowski, 2018-06-14:
If I don't use !1x but 1x the model starts saying "throttle active, throttle disabled"
When going from FM1 to FM0 it then says "throttle active".
This is good and normal behaviour upon selecting the model.

On a "reset flight" however it acts the same (saying "throttle active, throttle disabled") but when I go to FM0 from FM1 it does not say "throttle active"

The latter is incorrect for either !1x and 1x

kilrah, 2018-06-14:
> If I reset the model and later on press elevator down trim, it will go out of FM1 without playing the track.

That sequence if events was not made clear in the original description.

notlebowski, 2018-06-14:
Thanks for reopening the ticket.

Can you confirm the anomaly as well?

Wouldn't it be easier if "reset flight" would become the unloading of the model and loading it again?
Or would that cause unwanted side-effects, like breaking contact with the receiver?
I can imagine that the code would be simpler.

kilrah, 2018-06-15:
> Can you confirm the anomaly as well?

No. Please make it easier for us and explain how to go about with your model.
When starting the simu it starts in Safety mode. Long ele trim down switches to Normal. Further long down presses toggle between "Spoil" and "Normal" and I can't go back to "Safety". Don't have time to work through your logic to find out how to.

notlebowski, 2018-06-15:
I understand. I have an explanation in checklist, but lshems only made uploading of that possible a few days ago. The logic you already discovered is correct.
It's made for mode 4
To go back to safety you need to put left stick in corner right-down and the right stick to left-down.

To activate throttle there are 2 ways.
1 you already found out. For this the throttle needs to be idle.
There's a second one which is made for hand launch.
You can have the throttle full, but the motor will not run because of the safety.
Pressing elevator trim down for a long time starts a countdown (timer). At 0 the throttle is engaged.

Play track "throttle active" should always play when going from FM1 to FM0.
It doesn't play the track when the flight has been reset.

If you replace !FM1 with FM0 it will always play.

Everything works correct if you don't reset the flight (or break USB-connection)

notlebowski, 2018-06-15:
I have no idea what our mutual misunderstanding is. Hopefully I can explain it with a video.

It indeed works the same in simulator. I have already replaced "play track" with "play sound", so it can't be the wav.
I've even used another SF than SF1.

I have this same behaviour across multiple nightlies, on X9D, different simulators (windows, Mac).
I have no idea why you can't repeat it.

I therefore made a video of it.
Maybe that will clear it up.
If you have different behaviour doing the same I would be baffled.
Let's find out.

Video is still uploading... 
Comes in next post

notlebowski, 2018-06-15:
Dropbox was still busy, but it was already available with Google photos:
https://photos.app.goo.gl/KDZe61XiPXMg3TXEA

kilrah, 2018-06-15:
> I have no idea what our mutual misunderstanding is. Hopefully I can explain it with a video.

That you never mentioned this happened only if the Flight Reset is done while FM1 is already active.

When loading a model it starts in FM0 and the whole logic flips it into FM1. When doing a reset while already in FM1 it does not leave and reenter FM1. So no FM change is detected after reset, so the time counter for play delay is not initialized and the first change is muted.

notlebowski, 2018-06-15:
You are right in that it depends at which state the reset is done.
The Play track is always played if you reset it during FM0
But it's not the "!1x mute".

If the normal 1x is used it does exactly the same. No play track for going out of FM1

So...
SF1 !FM1 Play Track "thract" 1x

Being in FM1. Then reset
It plays "throttle active" then "throttle disabled".
Then press the trim and it will go from FM1 to FM0 without playing the track.

After a flight reset it somehow doesn't know that it is in FM1 even though it already played "throttle active and then throttle disabled"

notlebowski, 2018-06-15:
BTW...
I want to repeat again that I do not seek a solution for my model setup.
It's just that I feel worried when a flight reset does not behave exactly as a "new model selection".
What else it does maybe different...??

kilrah, 2018-06-15:
The mute comes from the "play delay" function which counts time between 2 FM changes and doesn't play a "flight mode exit" sound if they're too close. It also doesn't play if the time since last change is unknown, which is what happens after a reset.

notlebowski, 2018-06-15:
Ah... 
I interpreted "play delay" as the "first time mute".
I will change !FM1 to FM0 in my setup then. This ensures it's always played.

kilrah, 2018-06-15:
It is working as coded, whether it is as designed I don't know as it is likely years old. But can be easily changed without impact, so might as well do it.

notlebowski, 2018-06-16:
I tested 2.2.2N634 and can confirm it's now working as I was expecting it to.
It also plays after the reset.
Thanks for this.

commit ee4f8bda (2018-06-15): Don't reinitialise last flight mode on flight reset (Fix #5932)