opentx/opentx #5933: Companion can't reference newly made input. Needs to be restarted first

notlebowski, 2018-06-01:
When you create a new input in companion (nightly build 2 weeks ago, but probably all builds) you can't reference this input in a logical switch until you restart companion.
I believe a similar bug existed for logical switches, but that was fixed some time ago.

When creating this post I noticed this (last sentence)

https://www.rcgroups.com/forums/showpost.php?p=39657839&postco

kilrah, 2018-06-02:
You don't need to restart companion, just close/reopen the model.

There's already an issue open for all those propagation issues in companion.

notlebowski, 2018-06-02:
Sorry, wrote that down wrong.
Indeed, I just closed and reopened the model.

I didn't check github because I knew that the one involving LS was fixed and assumed that if there was one it was closed.

mpaperno, 2018-06-03:
@notlebowski  In the RCG post you said the problem is in the Special Functions panel. Not "in a logical switch" as you stated above... is that right?

In which case that's correct, and will be fixed by #5936 .

notlebowski, 2018-06-03:
@mpaperno  I was just about to post that I just tested it and you were right that it works in LS...
It was already 2 weeks ago that I encountered it and didn't have github then.
Good that you read my RCG-post....!!

I have another one which is more cosmetic than functional.
Just changed a name of a telemetry item and that change didn't get propagated across the setup unit model close/open.
I wouldn't mind if that would not be fixed
I leave it up to you

notlebowski, 2018-06-03:
Hi..
Not sure if we speak of the same thing and if it should work on N621
I just tested this on N621:

[image]
[image]
[image]
[image]

close/open model editing:

[image]

mpaperno, 2018-06-03:
Yep, we're on the same "page."  :)  Already worked for LSs, and confirmed fixed for SFs in #5936.

PR #5936 (2018-06-03): Fix remaining model editor sync issues with RawSource changes.
Fixes RawSource updates in Custom Function and Heli pages selectors.
Closes #4917 
Closes #5933
closed 2018-06-03 by bsongis