opentx/opentx #5934: Improve X- Lite startup

notlebowski, 2018-06-02:
When I received my X-Lite and tried to turn out on for the first time I thought I received a broken unit. I noticed the display coming up, a crackling sound and then a shutdown of the display making me let go of the power button.

I understand that this is part of a way to prevent the radio from turning on inadvertently when, for instance, some object is touching the power button for a moment.

I agree that this should be kept, but it would be better for both scenarios of the backlight came on immediately. In my scenario I wouldn't have let the power button go, because the backlight would be my feedback that all is well...

In case of an object touching the power button it would also be better if the backlight would then on as that event would then not stay unnoticed. You would be warned.

I hope you agree with this and that it's possible to show the backlight from the start.

3djc, 2018-06-02:
It is reallt strange you should say that, because I made myself the same observation : "strange, we forgot to turn backlight on".

Then I checked x7 and x9e, but no, we do not turn backlight on during power on push on those radio either

kilrah, 2018-06-02:
It's consistent indeed, but doesn't mean it can't be changed.

Pretty sure that it was initially chosen to avoid draining the battery in case something repeatedly presses on the button in a bag.
What's funny is that it's the 3rd radio that works this way but the first time people can't figure it out...

notlebowski, 2018-06-02:
Of course I'm a bit embarrassed that I didn't figure this out by myself. I must have seen too many devices that were brought for repair and behaved like that. It must have triggered me.

I still think that turning on is both more elegant and helps noticing when something is able to touch the power button.

notlebowski, 2018-06-13:
https://github.com/opentx/opentx/issues/5959

closed 2019-02-25 by notlebowski