opentx/opentx #5946: Connection has to skynavigator

domlouis, 2018-06-05:
I look for the way  to connect my Horus has to skynavigator to make some race for the GPS triangle.
I do not find an information on the subject.
The software to skynavigator being now paying I would like to be surethat it is possible before buying the software

schwabe, 2018-06-05:
We have no information about skynavigator. I think you should contact them directly.

domlouis, 2018-06-05:
Skynavigator is an android application

schwabe, 2018-06-05:
As I said, I never heard of it and the compability list on their website does not include OpenTX. For information I suggest to contact them.

domlouis, 2018-06-05:
OK vous n'en avez peut être pas entendue parlé , mais c'est le logiciel qui permet de gérer les compétition de triangle GPS pour planeur et reconnu dans toute l’Europe et US .
Les radio JETI font un forcing pour être compatible ...
si ça ne vous intéresse pas je trouve dommage de mettre en touche aussi vite , ça intéresse surement d'autre modéliste si vous ce n'est pas le cas ....
d’ailleurs je vois sur divers forum que beaucoup sont intéressé mais ne trouve pas de réponse .
Maintenant si c'est vous qui décidé qu’elle discipline de aéromodélisme sont intéressante OK !!!!!

kilrah, 2018-06-05:
Les développeurs d'OpenTX sont tous des volontaires qui travaillent dans leur temps libre sur ce qui leur plaìt/les intéresse. 

Le problème principal c'est qu'aucune info n'est a priori disponible même pour savoir comment le système fonctionne et déterminer ce qui serait nécessaire pour le supporter. 

Donc il faudrait d'abord nous fournir toutes les infos nécessaires. Je crois que personne dans l'équipe n'a même jamais entendu parler de la discipline...

Si assez de détails sont disponibles on peut réouvrir la requête et la mettre dans la liste en attente d'un développeur intéressé. Idéalement quelqu'un qui a un intérêt pour la discipline pourrait faire l'implémentation et nous la soumettre.

raphaelcoeffic, 2018-06-05:
@domlouis @kilrah La question semble aussi être: pourquoi est-ce que SkyNavigator ne travail pas avec FrSky pour pourvoir utiliser l'interface bluetooth présente dans la Horus? Sinon, pour quoi ne pas utiliser les systems standalone? (FlyMate et companie?) Et pour finir, pourquoi est-ce-que SkyNavigator n'offre pas une interface ouverte à tous? Il semble que la RCDroidbox supportait Frsky au début, mais faute de ventes suffisante, c'est destiné à Jeti seulement maintenant.

@domlouis Personne n'a décidé quoi que ce soit ;-) Et il est possible à tous d'implementer des fonctionnalités avec un logiciel libre. Il est même autorisé de payer un dévelopeur pour ceux qui ne peuvent le faire eux-même. Alors, s'il y a assez d'utilisateur potentiels, il devrait être possible de trouver quelqun pour faire le travail au final.

domlouis, 2018-06-05:
OK Ça j avais bien compris , et je ne remet pas du tout en cause le dévouement des développeurs  loin de la ....Bertrand étant un de mes collègues de club ....Je trouve juste très précipité de clore le sujet sans plus d'informations. ...

-------- Message d'origine --------

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [opentx/opentx] Connection has to skynavigator (#5946)",
"sections": [
{
"text": "",
"activityTitle": "**Andre Bernet**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@kilrah",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"opentx/opentx\",\n\"issueId\": 5946,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/opentx/opentx/issues/5946#issuecomment-394733301"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 342478516\n}"
}
],
"themeColor": "26292E"
}

raphaelcoeffic, 2018-06-05:
Je pense que le mieux serait de demander au dévelopeur de SkyNavigator ce qu'il en pense, et quelle serait la meilleur façon de faire avancer la chose ;-) Bertand à surement une petite idée comme cela pourrait être réalisé.

domlouis, 2018-06-05:
Vous avez raison c'est pourquoi j'ais deja envoyé des messages au developpeur , j'attend sa réponse.... 

----- Mail original -----

mpaperno, 2018-06-05:
There is almost no real information available about the software that I could easily find. The web site leaves a lot to be desired.  I can't even tell what it really means to "support it."  All the telemetry protocols used in OpenTX are open and there is lots of code to use as example.  Skynavigator should support these protocols perhaps, not the other way around...  or maybe pay others to integrate their proprietary (?) protocol.

One could at least be bothered to provide a link to what one is talking about...

domlouis, 2018-06-05:
oh la la stop n'en jeter plus laisser tomber ...... 
je ne voulais en aucun cas froissé qui que se soit ..... 
je voulais juste avoir des infos pour savoir si jetais le seul que ça intéressais .... 
je trouve juste dommage de racheter un tas de matériel alors que les infos a priori existe dans la horus , qu'elle a un module BT , et que l'application android existe déjà ..... 
pas grave je vais essayer de me débrouiller avec SKYNAVIGATOR .... 
MERCI pour tout ce que vous faite pour OpenTX 
amicalement 
dominique 

----- Mail original -----

bsongis, 2018-06-05:
Hello Dominique,

Personnellement le sujet m'intéresse et intéresse aussi d'autres devs, cf. notre chat interne. Donc on réouvre pour discuter de ce qu'on fait :)

Est-ce-que l'objectif est vraiment de s'interfacer avec SkyNavigator ? On a vraiment trop peu d'infos sur le site web. Est-ce-qu'on pourrait voir des copies d'écran, ou une description de ce qu'il faudrait faire pour atteindre le niveau de ce qu'ils proposent (pas au niveau doc ni site web - troll off).

Parce que imaginons une carte Google Map - ou plutôt OpenStreetMap (elles sont encore meilleures au moins si on regarde celles du Menez Hom) - et qu'on collait les 3 points GPS du triangle comme 3 points d'intérêt, puis qu'on y ajoutait notre planeur, est-ce-que cela serait un bon début ? Si oui il me semble qu'on doit avoir les compétences dans l'équipe. Peut-être même qu'on pourrait trouver un constructeur qui financerait un développement de ce genre d'application.

Bertrand

domlouis, 2018-06-06:
Hello Bertrand , 
la base c'est de pouvoir créer sur GOOGLE map des points (3 pour le triangle) , de les visualiser sur la carte , en traçant la forme géométrique (triangle si 3 points ) , et de visualiser la position et la trajectoire du planeur  , pour vérifier qu'il passe a l’extérieure des points ......
C'est le principe de base .
après il y a plein de fonctionnalité mais a voir...
voila une copie d'ecran 
[image]
[image]

bsongis, 2018-06-06:
Et si on arrive avec une autre application (qui fait la même chose, en libre), est-ce-que ce sera accepté par les jurys ? Ou est-ce-qu'on va devoir se perdre dans du lobying ?

domlouis, 2018-06-06:
Je ne sais pas , le souci c'est que Skynavigator est la seul application dispo , en plus elle était gratuite pendant tout le développement et maintenant elle est devenu payante

raphaelcoeffic, 2018-06-06:
@bsongis mon petit doigt m'a dit que SkyNav accepte NMEA par BT serial. Donc pas de problêmes, non?

Allerhopp, 2018-06-06:
hmm, strange language, allez hop ;)
NMEA over BT would allow also to use a lot of navigation programs, like locus pro, to follow the track of the plane, emulating a bluetooth GPS mouse.

bsongis, 2018-06-06:
NMEA on BT serial is not an issue, but Horus has BTLE, not BT serial

raphaelcoeffic, 2018-06-06:
That will require some custom hardware, taking the SPort data on one side, and ouputing NMEA on the other. That should be fairly simple to realize with an Arduino and HC-05 module.

Allerhopp, 2018-06-06:
Seems, that there was a converter, but not continued. @raphaelcoeffic : i do not have the skills, but i can do building/testing, if it is helpful.

https://www.rcgroups.com/forums/showpost.php?p=39531238&postcount=20

bsongis, 2018-06-06:
It seems to me that it is more simple to write an android app from scratch

bsongis, 2018-06-06:
Yes that would work

raphaelcoeffic, 2018-06-06:
Ok, we have all the necessary info now.

domlouis, 2018-09-10:
Hello 
 does anyone have any solutions?

raphaelcoeffic, 2018-09-20:
> does anyone have any solutions?

I delivered a LUA script that uses the new serial port capability, and SkyNav creator is currently working on it, basically integrating with his App.

The best would probably be to ask him directly, as we haven't exchanged a mail in a while now.

closed 2019-11-15 by stale[bot]