opentx/opentx #5953: X-lite stick PWM driver

MikeBland, 2018-06-08:
I'm porting ersky9x to the X-lite. I notice the openTx PWM driver code for the sticks uses a low interrupt priority of 10. I think this is too low. On my X-lite, I see a short pulse of just 52uS with a stick fully one way.
It is therefore necessary to ensure the stick interrupt is not delayed by this much time or a rising edge capture is lost. If, for example, per10ms() takes longer than this, and occurs at the "wrong" time a rising edge will be lost.
I also note that the period of the PWM signal is not constant. This means that a cetain pulse high time does not always correspond to the same stick position.
I'm using a high priority level (1) and measuring both the duty cycle and the period, and storing them from the interrupt.
I then calculate, when reading the analog values:
Stick position = (duty cycle) * 4095 / period.
This gives stick position values with a range of 0-4095, the same as the analog reading.

I'm doing NO averaging and get a VERY stable result, no noticeable jitter.

Mike

schwabe, 2018-06-08:
I am still waiting on my X-lite that I orderded online. When it arrives I can take a look at this. I think the rest of the OpenTX has preproduction x-lites which have analog sticks.

kilrah, 2018-06-09:
Preprod has PWM already, there is no Xlite with analog.

I believe the code actually comes from FrSky and was just included without much questioning beyond making sure it worked.

schwabe, 2018-06-25:
[image]

PWM periods from my preprod x-lite when moving sticks. Maximum difference is 14µs. Ignoring this would cause a 0,4% error. That doesn't sound like much but is actually 15 on the 4096 scale.

With stick centred and not moving:

Low to low:
[image]

High to high:
[image]

low to high:
[image]

Proposed by Mike:
[image]

(unit is wrong in the last one). Interstingly, even though the last one is calculated as floating point it falls into 6 distinctive bins.

Conclusion, Mike is right. Although on my unit I would still see a jitter of -2/+2.

schwabe, 2018-06-26:
@MikeBland although I agree with you in principle on the time spent in per10ms and not being able to capture all rising edges, I implemented a simple counter that checks if rising edge to rising edge is more than 6000. I can mange to increase this counter if I throw my left stick into the lower left corner, during which the readout of A1 actually shows 0000 but other than that my counter does not increase and everything stays the same wether I use priority 2 or 20 for the interrupt. Did I make some mistake in my thinking?

schwabe, 2018-07-14:
I am closing this for now as 968e7233e50a7ba48884896306dbe3fae570a951 should be good enough.

closed 2018-07-14 by schwabe