opentx/opentx #5956: SF reset telemetry does not work like other SF's

notlebowski, 2018-06-10:
Most SFs react on the switch **becoming** true. Play track, set timer, reset timer..
These all react on the "rise" of the switch.
The SF "reset telemetry" does NOT.
For me this was not expected behaviour.

I was trying to find out why my altitude was not working.
Because the GRX8 was recently in a crash it made me suspect a hardware failure.
It now turns out I'm constantly resetting the "altitude" because the condition (!FM1) stays TRUE.

I merely wanted a short Altitude reset when coming out of FM1. 
In my other models I always had a pulse available for that, but this model I created from scratch.

!FM1 reset TELE3:Alt ON

Now I realize this results in a constant reset of the altitude _(maybe even influencing the performance of the radio)_. During FM1 the throttle is set to -100 and prevents the motor from running. This was not what I intended.
I now will create an LS using edge that gives me a momentary TRUE upon FM1 becoming FALSE and use that as a trigger to reset the altitude.

I would like to know:

- Do you agree it is working as I describe?
- Do you agree it is not working like "Play track", "play value", "reset timer", "set timer", "haptic"?
- Do you agree it should work like those other SF's?

3djc, 2018-06-10:
It as been purposefully done this way I think during 2.2.0 RC. I cannot find again the issue that lead to that, but it is not mistake or bug, but more a design decision

kilrah, 2018-06-10:
> Do you agree it is working as I describe?

Yes and it is intended

> Do you agree it is not working like "Play track", "play value", "reset timer", "set timer", "haptic"?

No, all Resets except from Reset Flight behave the same.

closed 2018-06-10 by kilrah