opentx/opentx #5959: Backlight turns off on a Switch Warning.

Waldo-P-Barnstormer, 2018-06-11:
Taranis (non plus), on 2.2.2RC1
Backlight is turning off whenever a model is loaded and a Switch Warning occurs.
Clearing the warning turns the backlight back on.

To test, in Radio Settings have the Backlight Switch set to OFF.
Set a switch in Special Functions to turn ON the backlight. I use a Logical Switch based on the Right Slider position.
Have the backlight on and then load a model that will generate a switch warning. The light turns off.

In 2.2.1 the light is not turning off.
Jim

3djc, 2018-06-12:
We have reviewed the startup sequence for increased saftey, meaning some functions only start when all safety checks have been completed.

I guess we need to turn on backlight regarless of setings like we do for the other checks, but your LS wil remain disabled durung that period

Waldo-P-Barnstormer, 2018-06-12:
if reviewing the startup sequence would it be possible for the backlight to be turned on and left on while starting up.
There's an odd situation that's been there for some time. If a Splash Screen is set to be displayed, the backlight will turn off and then back on after the Splash Screen is over. If no time is set for the Splash Screen the backlight stays on. This occurs when using a Special Function to turn on/off the backlight.

3djc, 2018-06-13:
I'll have a look at that too tthen

notlebowski, 2018-06-13:
https://github.com/opentx/opentx/issues/5934

Can backlight be turned on during power-on sequence as well?
IMHO this is more intuitive and responsive.