opentx/opentx #5960: After Installing openTx op Horus X10s I got SD card Content Error

AidoSsy, 2018-06-11:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

After installing OpenTx 2.2.1 on Horus X10S and SD card content : sdcard-horus-2.2V0013, I got SD card Content corrupt error. I've tried to flash over again and reinstalling a new downloaded SD card content and the error apeared again. But after choosing FIX SD CARD CONTENT on first boot, everything seems to be working properly. IS IT?

raphaelcoeffic, 2018-06-11:
The SD content that you downloaded and installed does not contain the model and radio settings, which is what the error is about. This created on first boot and is absolutely normal.

closed 2018-06-11 by raphaelcoeffic

AidoSsy, 2018-06-11:
Thank you. This should be written down on the instructions.