opentx/opentx #5968: Updted wiki needed: X9D+ serial port connect to bluetooth

davidngrc, 2018-06-14:
This issue was discussed here, but not detail enough.
https://github.com/opentx/opentx/issues/3465

could someone please update the wiki, and detail the wire connection.
https://github.com/opentx/opentx/wiki/Taranis-I-O-ports

Thank you.

3djc, 2018-06-14:
The wiki mostly toward software developpment, and therefore includes some hardware info needed for that purpose, but is not intended to be an 'everything you can do with a (place an OpenTx supported radio model here)' database.

You question is probably best suited for support places like rcgroup or openrcforum

kilrah, 2018-06-14:
That page is a description of the interface on the radio, not a manual about doing specific things with it.

closed 2018-06-14 by kilrah