opentx/opentx #5971: openTx 2.2.2RC1 R9M latest firmware no 200mw telemetry

SalterFernandes, 2018-06-15:
Im using openTx 2.2.2RC1
R9M with 180428 LBT firmware 
R9 Slim with 180425 LBT firmware

Cant get 200mw with telemetry option, just on the 25mw.

3djc, 2018-06-15:
Radio type ?

As per Frsky release notes:
Add 16 channels with telemetry and 16 channels **without** telemetry (200mw power) modes.

3djc, 2018-06-15:
I jhust confirmed with frsky, it is : Add 16 channels with telemetry(25mw) and 16 channels without telemetry (200mw power) modes.

closed 2018-06-15 by 3djc

SalterFernandes, 2018-06-15:
Sorry, forgot the radio... its an QX7

Well sorry then for the false report, but they should be clear on the info, i was reading 200mw with and without Telemetry.

3djc, 2018-06-15:
Indeed, I have suggested they update their wordiing