opentx/opentx #5979: Support Jumper T8SG V2.0 Plus

tgreer, 2018-06-20:
What's involved in supporting another TX? This is a nice format, has a module bay and hall gimbals.

3djc, 2018-06-20:
It does not seem to have sdcard, which makes it a bad candidate to start with, imho

schwabe, 2018-06-20:
It is also supported by DeviationTx already, another OSS project. It seem to be a [Devo10](https://www.deviationtx.com/wiki/hardware/devo10) variant (or clone from a schematics perspective)

The biggest problems for porting OpenTX are:

- OpenTX is rather large, that chip has "only" 256kB of flash
- no sd card
- OpenTX has no own support for speaking radio protocols. So that would need to be ported/implemented from Deviation

While possible, someone needs to do all the effort to do this port. And then you need a clear reason why you want OpenTX instead of DeviationTX on that TX. At the moment I cannot see any.

tgreer, 2018-06-20:
Understandable if it's not a great candidate. I'd prefer opentx on it for a number of reasons. 1) familiarity, 2) CRSF support, 3) Lua support. It's a really nice format in the hands, which suits me much better. I find the X9D too big and the x-lite doesn't sit well with me (xbox controller style). I shall keep hunting for a TX which ticks the boxes for me and opentx!

closed 2018-06-20 by tgreer

schwabe, 2018-06-20:
CSRF support is probbly something that can be implemented in Deviation. For the LUA support, that won't work on that remote anyway since it has only 48k of total RAM and the current BF script even crashes with out of memory with > 60 kB free memory.

You probably need a leaner bf pid implementation on Devatition (probably in C) which is totally doable.

tgreer, 2018-06-20:
It's actually for Telem screen Lua stuff rather than BF PIDs :) But yes
point taken.

3djc, 2018-10-04:
Certainly NOT

There is no OpenTx support for this radio, nor is there any planned. They are using a forked version of OpenTx, that they have no legal right calling OpenTx.

They will have to maintain their code by themselves (which is unlikely they will do), and firmware are not going to be available using OpenTx companion.

Don't be fooled by what some asian shop are advertising ....

dronevate, 2021-05-04:
Had to eat thos words eh?