opentx/opentx #5980: Unexpected behaviour "Reset Flight"

pinkywafer, 2018-06-20:
I have previously (in OpenTX 2.1) used a special function to "reset flight" on arm.  On 2.2, this is now not possible. (Not sure when this changed - but just loaded 2.2.2RC1)
this issue is related (I think) to #4073

Simplified version of my setup:
Switch SFup for throttle override (this is used for a -100 mix to override the throttle)

L01 Edge  SFdown 0.00 (instant)

SF1 L01 Reset Flight

This now causes the switch warning to warn me that SF is up, display preflight checklist, and also gives a throttle warning if the throttle is open.
Previously, the data (Timers + Telemetry values) were reset, but now, we're almost back to reloading the model.  Obviously, I would have to use up to 4 functions to serve the same purpose (3x timers + telemetry).
However the bigger problem for me is when it does reset, the module is NOT power cycled (unlike when selecting a new model), so the alarms are free to go off with throttle open in my example.

If reset flight is going to return to just loaded model state, can it also cycle the power to the module
AND, can we have a separate function to reset all (timers & telemetry) so as to not require 4 Special functions to do the same job.  Personally, I'd prefer the old "Reset Flight"
If needed, a "Reset Model" to return to just loaded state could be added though I can't see why itwould be needed.

pinkywafer, 2018-06-20:
just spotted this being discussed before in #3422  
If this is staying as it is, can we please have a new function to just reset all timers + telemetry and have this function behave exactly as when loading the model?

kilrah, 2018-06-21:
The current situation is intentional becasue powering the module off would trigger failsafe on the receiver that we know is powered and which may be improperly configured.

If your model is in a safe position to execute "Reset Flight" then it should not be a problem.

3djc, 2018-06-21:
Yes, if you run the function, then it is fair to assume that is is safe to do at that time. Turning off module is never safe (I would really dream that every modelers would setup it's failsafe correctly, but we know it is not the case)

pinkywafer, 2018-06-21:
Point taken regarding cycling the module.
When I'm flying on private land, and only want to change battery and fly again, I leave transmitter on, change battery, quick preflight and launch straight away.  I don't want to forget to reset all my timers etc, so have previously had 'reset flight' linked to the throttle activation. I have always found that function to reset all timers + telemetry has always been useful as it was for this application.

lshems, 2018-06-27:
I have a problem with accessing the reset flight from the long press enter if it will trigger my switch warnings, including sdcard warning values.

If I do not enter into the menu deliberately, I don't want such checks to be performed 'by accident', touching some other buttons.

I have the feeling opentx is becoming more and more functionally oriented, which is in conflict with the idea that opentx can be used to control anything you like.

The scenarios discussed are fine for some, but not for all.

What happens when all these switch warnings are issued after a flight reset? Is control blocked as it is with the checks on a model change?

It's absolutely unclear to me what these warnings would do, and I cannot imagine it is the same as on model change. So at least there must be some indication to what these warnings actually want you to do.

pinkywafer, 2018-06-27:
If the "Reset Flight" is being performed by the long press enter method, it would be fairly hard to do this accidentally, so fair enough,
BUT with this function available as a Special function, so able to be activated by a switch (or logical switch), it would definitely be possible for this to be activated in flight, at a time when switch / throttle position warnings etc and pre-flight checks are definitely not wanted, thus to achieve the same effect as previously (resetting all timers + telemetry) would require 4 alternative SFs

lshems, 2018-06-27:
I do not agree. Long press enter to reset a timer, and I miss one press. Bang, reset shit all over the place.

Reset in this way needs to be restricted to model selection screen.

How much discussion has been had on accessing change model on one click, because if the apparent danger, and e would allow this?

Just because you don't want four SF ? You have 64!!!!

mpaperno, 2018-06-27:
I agree that it sounds dangerous, especially since a user could be in that menu just to reset a timer or telemetry and easily choose the wrong line while concentrating on the model.

I also think it was a poor decision to change the default behavior after the feature has been well established. 

> Perhaps and option in model configuration, that would (when enabled) check switches position after flight reset.

pinkywafer, 2018-06-27:
When I first noticed this, yes I was annoyed that I may need 4SFs, but that is because I would have to change to 4 SFs for every model I have this set up on (most).
I do agree that this is not a good change from a safety point of view especially since the behaviour of reset flight has long been established

lshems, 2018-06-27:
I would be happy to write you a lua script to add the four SF. Send me an
example epee by pm,or post it here.

pinkywafer, 2018-06-28:
Thanks for the offer,  but I have completed the changes for now. Too bad I didn't think of writing a script!  (Especially as I had to modify my 'initial setup script) but the exercise did give me a chance to get some standardisation done as well (some of my setups were very different to my current way of doing things!)

closed 2019-11-15 by stale[bot]