opentx/opentx #5984: the function "SetFailsafe" is resetting GV1 to "0" in activated flight mode

ParkerEde, 2018-06-21:
Hi,
I've found the following issue with my Horus X10S and OpenTX 2.2.1:
Global function: switch "SA (up)" -> SetFaisafe internal
In the modelsetup I use GV1 in three flight modes for the weight for the input of ailerons.
If I set the switch "SA" to (up) position the failsafe values will set correctly to the failsafesetup, but GV1 will also set to "0" in the activated flight mode.

regards
Ralf

3djc, 2018-06-22:
Please post your otx

ParkerEde, 2018-06-22:
[X10S_2.2.2RC2-GV1-resetting-issue.zip](https://github.com/opentx/opentx/files/2126376/X10S_2.2.2RC2-GV1-resetting-issue.zip)

ParkerEde, 2018-06-22:
[image]

[image]

closed 2018-06-22 by 3djc

ParkerEde, 2018-06-23:
recheck in 2.2.2N641 -> okay
Thank you

commit 4d126dd9 (2018-06-22): Fix setFailafe SF
Fixes #5984