opentx/opentx #5990: Companion display error for glabal functions

Storchy1, 2018-06-25:
OpenTX 2.2.2 RC2.

I Have made a Global Function to set GV9 to zero see GF4 
[image]

in the companion this function is not displayed correct
[image]

Please have a look

Storchy1, 2018-06-25:
I don't how i have made this, perhaps in former version it was allowed. I have now deleted GF4 and i am not able to create the function once more. You are right, Adjust is not available.
If i find a way to create the function once more i will write you
Thank you very much

kilrah, 2018-06-25:
Yeah it's possible the check was missing in a previous version. Or I found out you could cheat by setting up an Adjust GVAR model special function then copy/paste it into global functions. That's now no longer possible.

commit c3b23fcc (2018-06-25): Prevent users from cheating (fix #5990)