opentx/opentx #6005: Vario tones don't work if you switch receivers in model

gquiring, 2018-06-30:
If you build a model with a S8R receiver (or other receivers) and then switch receivers to a G-RX8 and rebind the vario tones don't work correctly. Deleting all sensors and rediscover does not fix the issue.  

It's been discussed on RCG and seems to be an issue with any hardware.  You have to build a new model to get the vario to work.  Something is being left over from the old receiver when you switch them.

I enclosed a sound file of what the vario sounds like.  
[Vario.zip](https://github.com/opentx/opentx/files/2151303/Vario.zip)

Here is the post on RCG by another user:
https://www.rcgroups.com/forums/showpost.php?p=39775049&postcount=2018

I am using:
OpenTX 2.2.1
Horus X10S

3djc, 2018-06-30:
For Vario to work, you need it:
- ben seen as a sensor, which you did by rediscovering sensors
- set a vario source in the telemetry screen, which I guess you forgot

Tested and working fine here

closed 2018-06-30 by 3djc

gquiring, 2018-06-30:
It is set as the vario source on the telemetry screen.  Do you understand that it works perfectly fine unless you have an existing model with a different RX.  Switch out the RX in the model with the GRX8, bind, delete all sensors, rediscover sensors, set sensors and the tones do not work as the zip file I enclosed shows.  It's been reported by several of us on the forums, this is a bug.

gquiring, 2018-06-30:
Did you actually test by switching receivers 3djc?

3djc, 2018-06-30:
Yes tested by switching from a x8r+alt to g-rx8

gquiring, 2018-06-30:
Here is the model. 

[model3.zip](https://github.com/opentx/opentx/files/2151682/model3.zip)

raphaelcoeffic, 2018-06-30:
As I expected, and other people have told you on rcgroups: the vario source is wrong! You are using "Alt", which is the absolute altitude, whereby it should be "VSpd".

Here is what I see on your model:
[image]

3djc, 2018-06-30:
You source should not be alt, but vspd.....

gquiring, 2018-06-30:
I will try it, FYI Raphelcoeffic, I would like to know where on RCG they told me to use VSPD?  The only hint I got was to create a new model by member Barry_Cole.

Thank you 3djc, I misunderstood what you were saying with the sensor.

raphaelcoeffic, 2018-06-30:
@gquiring everyone said you need to set the vario source properly ;-) As you did not provide any more info, that's all they could answer.