opentx/opentx #6007: Sounds files not always released by simulated firmware

mpaperno, 2018-06-30:
In Simulator (launched from Companion), when playing a track as "background music" the file is not closed/released, even after simulator exit.  The open file handle(s) can be observed and belong(s) to the Companion process.  The file handles remain even if the track is re-started (not paused), and a new one is opened, so they can accumulate. 

I also observed this when trying to play a track (as "Play Track" SF) which wouldn't play in simulator at all (it was a hi-fi wav I randomly grabbed from my music collection so I'm assuming it was the wrong format or something).  There was no visible error.  But the file remained open/locked afterwards, like with the background music.

The tracks are then unplayable in Companion (and I presume other apps) because the files are locked.  It's unclear to me how SDL can open multiple handles though.  They're released when Companion is closed.

I've only observed the locking behavior on Windows (seems to not lock on Mac), and only examined the actual open handles on Windows.

"Play Track" with a valid/playable file properly closes and releases file handles.

Related: #6000

mpaperno, 2018-06-30:
Confirmed that file handles also remain open in Linux. Though the files aren't locked (eg. for playback in Companion), so I guess it's mostly "cosmetic" there.

raphaelcoeffic, 2018-06-30:
Until it hits the max open file limit of 1024 by default ;-)