opentx/opentx #6017: OpenTX 2.2.2 RC3 lines

marcogpw, 2018-07-05:
Just upgraded Companion on my PC from 2.2.2 RC2 to 2.2.2 RC3 and wrote the new firmware to my X12S Horus (also running on OpenTX 2.2.2 RC2), after the startup splash screen I got the failsafe Warning screen with some lines through the text, also on various other screens in normal operation I got the lines (see screen shots)
[image]
[image]
[image]

I tried to remedy it by reverting back to firmware 2.2.2 RC2, but with same results (companion on PC still running on RC3).
Only after rolling back to companion RC2 on PC and writing firmware RC2 again to my Horus the lines were gone and everything was normal again.

Marco

3djc, 2018-07-05:
I'm absolutely unable to duplicate that, or to understand what could be causing that with the very limited amount of changes that went into rc3. Also, flashing with companion rc2 or companion rc3 doesn't make any diffrence, has this is done by the party programs that havent changed.

I would like some precision:
- what where your firmware options ?
- please post full otx (radio +model)
- you said you went back to rc2, did you actually save an rc2 firmware, or did you dl companion rc2 and assumed it will download an rc2 firmware (it will not, even rc2 does give you rc3 firmware as of today)
- what kind of module if any is in the radio bay, and anything you could tell us that would make you radio 'different'

marcogpw, 2018-07-05:
My firmware option are:

Noheli, Lua and Luac
My Radio is a normal production model X12S Horus and I redownloaded the RC2 Companion anew and reinstalled it, then I wrote the RC2 firmware that I had saved on my PC
There is no module what so ever in the radio bay. The only thing I changed are 2 swithes I changed to a different type (pull out safeties to avoid accidental operation) and one of the 3 position switches now has a spring loaded momentary position on one of the 3 positions) other than that it's standard and hasn't given me any problems. The build date of the transmitter is December 3rd 2016 and as of January 2017 I am running OpenTX on it.

3djc, 2018-07-05:
Yes, but you mention flashing saved RC2 with RC3 CPN and that failing, and flashing the same firmware with RC2 CPN being ok. I cannot think of any logical explanation for that.

Tried the same firmware options without issues

marcogpw, 2018-07-06:
I tried RC4, same results: also the same lines as with RC3, reverted back to RC2 (companion and firmware) and got some lines as well (less than with RC3 and RC4). So I thought it could be a hardware problem, so opened up the X12S and checked all connection to and from the display and data transfer/SD card board by taking the ribbon cables out and putting them back again.
Loaded RC4 again: same results, lines again. Reverted back to RC2, still lines ! Tried reverting back to RC1 companion and firmware (had all firmware versions on my PC still), and to my surprise all lines are gone now ... the only thing I changed were the Companion version on the PC and the firmware on the X12S, nothing more. So I expect some kind of software - hardware issue in my setup, since you could not reproduce the issues I have.
Are there any other users with the same problems ?

bsongis, 2018-07-07:
Did you see the problem occurs only on DBLSIZE font? That's weird!

bsongis, 2018-07-07:
I should even say ALWAYS on DBLSIZE font!