opentx/opentx #6024: In simulation - missing sliders

Hudla, 2018-07-08:
hello,

The Companion 2.2.1 for FrSky X9D + radio is not a side slider display in the simulation.
Thank you in advance

Robert

**General information**
 - OpenTX version 2.2.1
 - Hardware Frsky X9D+ 
 - OpenTX Companion 2.2.1

[image]

mpaperno, 2018-07-08:
Do you have the sliders configured in Radio Settings for this file? 

[image]
 (Image is of X9E, there would only be two slots with that option on an X9D)

Hudla, 2018-07-09:
Thank you very much for your advice.
I'm a little shy that I did not get it.
Next time I will be smarter ... 
:-)

mpaperno, 2018-07-09:
No problem, you/re not the first!

closed 2018-07-09 by mpaperno