opentx/opentx #6026: Please fix model.setCurve()

Miami-Mike, 2018-07-09:
There are two issues with model.setCurve(), one with creating a Custom X curve and the other with creating a Fixed X curve.

The issue with creating a Custom X curve is that an "X" value of zero results in a value of -100%, but any  value above that, from 1 to 100, plots a point to the right of the Y axis, and any negative value results in a return of "5", meaning "x value not monotonically increasing." There appears to be no way to plot negative X values other than X = -100%.

The issue with creating a Fixed X curve is that Y values returned from model.getCurve() are indexed starting with zero, yet model.setCurve() requires Y values indexed starting with one. That means that copying a curve from one slot ("oldcurve") to another ("newcurve") requires incrementing the index like this:

for i = 1, curvepoints do
	newcurve.y[i] = oldcurve.y[i - 1]
end

This second issue is interfering with a project I'm working on because I don't know how you'll decide to fix it. Whether you fix model.getCurve() or model.setCurve(), some scripts will be broken. 

For further details please see my post on RCCGroups:

https://www.rcgroups.com/forums/showpost.php?p=39907354&postcount=2656

Miami-Mike, 2018-07-11:
There's another discrepancy between model.getCurve() and model.setCurve() that causes difficulty when reading a curve, modifying it, and writing it back. The problem is that in model.getCurve() the smooth parameter is boolean, but in model.setCurve() it has to be a number, either 0 for not smooth or 1 for smooth. You probably confused it with the type parameter in model.getCurve(), which is 0 for Fixed X and 1 for Custom X.

The result is that when trying to write back the parameters obtained with model.getCurve(), you not only have to adjust the y values, you also have to correct the smooth parameter, else you get "bad argument #-1 to 'setCurve' (number expected, got boolean)."

lshems, 2018-07-11:
In my humble opinion it would be very bad if you can't use setcurve after a getcurve without changing something.

That would defy any logic.

Instead of fixing it the dirty way with a parameter, I would suggest to expand all 'set' functions with a return of the 'get'.

So to get a certain setting, you would use 'set' without any parameters.

If you set one parameter with 'set', you would get returned the complete set of the current 'get' function.

Any Lua variable containing the parameters can then be updated with one command:

Getting the current settings:
Parameters =setsomething ([])
Parameters will get filled with current settings [a=1,b=2]

Changing only the a parameter:
Parameters= setsomething ([a=parameters.a+1])

This would not change any existing logic, and could in future versions eliminate half of the functions. (All the get functions become obsolete)

schwabe, 2018-07-11:
@lshems that sound like very stupid idea. To have set functions that are really get functions

lshems, 2018-07-12:
It is very common practice to have a function to change something, and get the changed results as a return argument.

It is mostly used in combination with range chacks on the entry parameters.

Suppose the input range of parameters a is -100 to 100, you can very easily use this behaviour to make incremented for your setting, that won't get incremented when hitting the limit. Returning the resulting value would inform you about the actual value, that would be the range limit if overriding.

In that way you don't need to keep a register of limits for the parameters of your functions, or use a get after every set.

I have made a lua function for every model set/get combi that does just that, combine the set and the get, and do the range check. It saves me tons of lines of code.

But hey, the idea sounds stupid, so please do not implement it.

schwabe, 2018-07-13:
I will try to get the minimal changes to make the function usuable into 2.2.2 (bool vs int and -100 instead 0 fix) any major function overhaul needs to wait for the next version.

@lshems I don't think that set isntead of get/set is a good idea. But if we make such a change it has to be in a major version like 2.3 and then cosistent for all functions.

Miami-Mike, 2018-07-13:
Item #1: Regarding creating a Custom X curve, I hope you'll make sure that it'll be possible to create any curve that OpenTX allows you to manually create, and that should include allowing two consecutive X coordinates to be equal. That's not only allowed by OpenTX, according to Wikipedia it's consistent with the stated requirement that "x values must be monotonically increasing."

From [Wikipedia](https://en.wikipedia.org/wiki/Monotonic_function):

>In calculus, a function defined on a subset of the real numbers with real values is called monotonic if and only if it is either entirely non-increasing, or entirely non-decreasing. That is, a function that increases monotonically does not exclusively have to increase, it simply must not decrease.

The reason this matters is that for my project I plan to use what I call "stair curves", and I'd like to be able to generate them with Lua. Here's an example:

[image]

Item #2: Regarding the issue of the _smooth_ parameter, do you intend to change model.getCurve() or model.setCurve()? If model.setCurve(), will your change generate errors from code that currently works?

Item #3: Also, if you can, please look into [my request to make the trim switches accessible to Lua](https://github.com/opentx/opentx/issues/6040).

schwabe, 2018-07-13:
For item 2. I changed the parameter in my PR to be a bool in setCurve but 0/1 is still accepted but not documented for legacy. 

For number #1 that should already work. The code only bails out if the previous point is larger.
```
    // Check that x values are increasing
    for (int i = 1; i < numPoints; i++) {
      if (xPoints[i - 1] > xPoints[i]) {
        lua_pushinteger(L, 5);
        return 1;
      }
    }
```

Note, it says monotonically increasing not strictly monotonically. ;)

For item 3, lets not mix issues here.

3djc, 2018-07-13:
Just as a side comment, no now features will be going to 2.2.2, all new ones will be earliest in 2.3

closed 2018-07-13 by bsongis

Miami-Mike, 2018-07-16:
Great work, guys! I tested model.setCurve() with OpenTX 2.2.2 RC5 and it successfully writes or copies Custom X curves, including the "stair curve" that I illustrated above. It also accepts the smooth parameter as either a boolean, or a zero or one.

Thanks!

schwabe, 2018-07-16:
@Miami-Mike thanks for testing and confirming that the fix works as intended.

commit c99dc375 (2018-07-13): Fix setting x values < 0 for custom curves (fixes #6026)

Also accept bool for smooth parameter as the documentation suggest
commit 5ee3825a (2018-07-13): Merge pull request #6041 from opentx/schwabe/fix_setcurves_6026

Fix setting x values < 0 for custom curvs (fixes #6026)