opentx/opentx #6028: opentx 2.2 + on turnigy th9x

norwich2817, 2018-07-10:
Is there any plan to add compatibility for the turnigy th-9x in future releases of opentx?  Or is there some other way to get it to work?

norwich2817, 2018-07-11:
I know that I need to stay on 2.1 for now, I'm curious if there is a plan to add support for 2.2 in the future.  I ask because I want to use one of the I range x multi protocol modules but they require a 2.2 feature to work.  If there will never be support then I am considering upgrading to a different radio

3djc, 2018-07-11:
Sorry, did not fully understand your question the first time. But no, we have no plan for 're adding' support for AVR in future version (quite the contrary, we are removing AVR specific code to improve the code for the rest of the radios). Simply, there is no room on those radio to add the new code (2.1 for AVR is mostly 2.0 already, because there was no room left)