opentx/opentx #6029: Scripts called by Global Functions run for all models whether or not Global Functions is checked.

Miami-Mike, 2018-07-10:
Observed in Companion, nightly build 2.2.2N655, set for X9D+ and again set for X-Lite. Not confirmed in any radio.

Set any function script to be activated as a Global Function and it will work with any model selected, no matter whether or not the Global Functions box is checked on that model's Setup page. 

Example:

Name this simple script "99mph.lua", store it in SCRIPTS/FUNCTIONS, and set it to run as a Global Function with switch set to "ON".

local function run_func()
playNumber(99, 8)
end
return {run=run_func}
 
Simulate any model in memory and it will repeat "99 miles per hour" whether or not the Global Functions box is checked.

Miami-Mike, 2018-07-16:
Now, as of 2.2.2RC5, there are at least some cases where a script called by a special function won't load unless global functions are enabled.

kilrah, 2018-07-16:
Example and way to reproduce?

Miami-Mike, 2018-07-16:
I learned a little more about why the problem doesn't always show up. Here's an example script named _test.lua_:

    local n = 1
    local function run_func()
        n = n + 1
        print(n)
    end
    return {run=run_func}

Set up two special functions to run it:

SF1 SA↑ Play Script test
SF2 SA↓ Play Script test

Watch Debug Output. If Global Functions is checked then it'll work with either SA↑ or SA↓, but with Global Functions not checked, it'll only work with SA↑, not SA↓.

kilrah, 2018-07-16:
It breaks something indeed. Can't understand what so I have to revert the fix and someone else will hopefully look at it.

lshems, 2019-08-28:
dangerous??

commit b50826bc (2018-07-14): Do not load global function scripts if global functions are disabled in model settings (fix #6029)
commit f1dafb0d (2018-07-16): Revert "Do not load global function scripts if global functions are disabled in model settings (fix #6029)"

This reverts commit b50826bc74ee627849d10eb2b6267b0361e8edae.
commit 6684a75d (2019-09-10): Fixes #6029
closed 2019-09-10 by bsongis