opentx/opentx #6030: Companion Edit Radio Settings issue

marcogpw, 2018-07-10:
In the Edit Radio Settings tab of Companion (RC4 version on PC) at the GPS Coordinates, there are 2 options, like on the radio setup menu on the radio. In Companion the options are hh° (N/S) mm' ss".dd this is not correct for coordinate systems ... in the radio it states DMS for Degrees, Minutes and Seconds, somehow in Companion it is stated as Hours, Minutes and Seconds. 
I was setting up my XLite and saw it in the edit radio settings tab.
Minor flaw ...

raphaelcoeffic, 2019-08-26:
@elecpower Hey! It seems the necessary changes have already been implemented and merged. Can you please confirm that we can close this issue?

kilrah, 2019-08-26:
Yeah it was just not auto-closed becasue the ref was missing in the PR

PR #6300 (2019-05-18): Align Companion gps coords format choices to radio
Fixes #6030
closed 2019-08-26 by kilrah