opentx/opentx #6031: Add LUA script screen on X10S

IcemanDZ, 2018-07-11:
Hi, I just got the frsky X10S, and I flashed the OPENTX 2.2.2 RC4, however I find that the custom scripts is not working on the radio, we need go the SD file system and manually execute, which is very complicated, I thought if I hold the tele button on the radio,  the LUA script will shown on the screen, but it's not, it shows the main view setting, I think it‘s better to show the lua script better than main view settings, the main view settings can change to another button.

mpaperno, 2018-07-11:
There are no "telemetry screens" on the color radios. They use the "widgets" concept instead.  You can modify which widgets are displayed, and number of screens & their layouts, from that "main view" setting you found.  A few widgets are built-in, others written in Lua can be added on SD card.

The 9X scripts pretty much wouldn't work anyway because of the different screen. But should not be hard to adapt.

closed 2018-07-11 by mpaperno

lshems, 2018-07-11:
Check my blog on rcgroups on info how to convert telemetry scripts to widgets.

L shems.