opentx/opentx #6040: Request: Please make the trim switches available to Lua.

Miami-Mike, 2018-07-13:
Lua can read trim-ail, trim-ele, trim-rud, and trim-thr via the getValue() function, but it would be nice to have a way to directly read the states of those switches even when they're disabled. I don't want to read the trim values, I want to disable a trim on the Flight Modes tab and use the trim switches for something else.

StefanKellerAC, 2018-08-11:
yes, that would be helpful.
additionally it would be good to be able to set individual trims with the special functions and with Lua...

Miami-Mike, 2018-08-11:
I think the developers want us to keep each individual request or issue in its own separate thread because of the way they're managed here.

StefanKellerAC, 2018-08-12:
done. #6098
in addition to your request: 
when a trim is disabled the announcements "maximum trim reached" and "trim centered"  should not be spoken.

Miami-Mike, 2018-08-12:
I can't figure out what you're talking about. When a trim is disabled those announcements ****aren't**** spoken. In fact, nothing visible or audible happens when a disabled trim switch is actuated, unless a special function is set up to be activated by it. I tried it in Companion 2.1.9 and 2.2.2.

At any rate, if you really think the issue exists then you should start another thread about it like you did before.

RC-SOAR, 2019-03-27:
I'd like to add further weight to MiamiMike's request (which seems to have got lost in the fog): a Lua function please, to return the raw state of a trim lever:

**getRawtrim ([name|id])**
returns -100 (down), 0(centre) or 100(up)

Yes there's a workaround (define 3 SF's to update a GV, and query the GV in Lua), but it's messy and it uses up a precious GV. Having a dedicated function would make it much easier to use the trim levers as general purpose 3-pos switches, very useful for applications which need to repurpose trims.

kilrah, 2019-03-27:
Um the trim keys are switches, so you should be able to get them as that?

RC-SOAR, 2019-03-27:
That's what I hoped, but **getValue("trim-ele")** only returns a value if the trim is enabled in the FM menu. 

If the elevator trim is disabled, then **getValue ("trim-ele")** returns zero regardless of the state (up/down/middle).

kilrah, 2019-03-27:
OK, this returns the trim register, seems there's indee dno way to access just the keys.

lshems, 2019-03-27:
Yes please.

RC-SOAR, 2019-03-28:
Shame if it can't be done, it would turn OpenTx into the complete Jeti killer. Users could easily retrofit stuff like dynamic trims, not possible with any current radio AFAIK. Currently, the workaround method (1 GV + 3SF's) isn't possible without extra work and compromise.

closed 2019-12-23 by stale[bot]