opentx/opentx #6042: OTX 2.2.2RC4, RAS/SWR

Nihonski, 2018-07-13:
I still would like to indicate SWR as a telemetry value, so I tried to a create a sensor input manually with ID F105 and instance 25, however there is no value indicated.
When check the S.Port mirror on my X9D+, the data string for SWR is still there, but why I cannot set it up for indication on the screen?

schwabe, 2018-07-13:
Read the news of OpenTX 2.2. SWR is not available as telemetry anymore, only as RAS under HW.

closed 2018-07-13 by schwabe

schwabe, 2018-07-17:
you can still see the SWR/RAS under the hw analogs.

3djc, 2018-07-17:
And obviously, should RAS get into error level, the radio will tell you also.

schwabe, 2018-07-17:
But too many users thought RAS good means no problem. But it is only RAS bad means that you have a problem.

Nihonski, 2018-07-17:
I know, but its very troublesome to check the value by this way before every flight.
Moreover there is no way to set the threshold of the warning.
If I cant see, I would at least decide by myself, when I get a warning.
Pitty, the SWR HW is there, but cant be used easily.

schwabe, 2018-07-17:
@Nihonski that would describe is exactly what is the problematic behaviour. If you are checking RAS before every flight, you are trusting the RAS value. Experience shows that it cannot be trusted unless it is high.

3djc, 2018-07-17:
Thats the whole point ! It is NOT by any mean an analog read of swr. It is a strange pass/fail logic with analog output. If they had know from start the mess it would make, they probably would have put just an additional transistor to properly switch 0/1, but they wanted to save one I guess

kilrah, 2018-07-17:
It's accessible to lua scripts I believe.

Nihonski, 2018-07-17:
I think the problem is, that the analog signals of the forward and return RF-detetectors are not properly filtered, neither by HW, nor by SW. The detetcor signal has quite some ripple, and sporadic drop outs because of LBT. Any SW filtering needs to by synchronised with the tx/rx cycle.
I could fix that, but would like to adjust the warning threshold.

schwabe, 2018-07-17:
@Nihonski You are free to do so. And as @kilrah said the value is accessible via LUA. If you do your modifcations to OpenTX itself you can also modify the settings there. 

But I would suggest not spending to much time on it.

kilrah, 2018-07-17:
There is also no calibration at all nor any effort to keep tolerances of the detector. It was never designed to provide more than hopefully a pass/fail information.

Nihonski, 2018-07-17:
Ok, thank your feedback and spending your time. Closing the issue is fine with me.