opentx/opentx #6045: Opentx-9x-2.0.20 --14-07-2018 EEPR 216-0

nino165, 2018-07-14:
Un saluto a tutti
Ho scaricato ed installato la 2.0.20 nella mia turnigy 9x
all'accensione della radio il primo avviso è il motore non in posizione la calibrazione è fatta e lo stick del gas e tutto chiuso.mi consiglia di premere un tasto.
Dopo questo avviso arriva il secondo CONTROLLI THR e AIL sono selezionati come se lo swich sia in posizione da 0 a 1, quindi attivi. Mi dice di premere un tasto.
Premuto mi riporta dinuovo a calibrare gli stick per la quinta volta.
Qualcuno mi sa dire cosa fare?

3djc, 2018-07-14:
Please only english here.

closed 2018-07-14 by 3djc

nino165, 2018-07-14:
Hello everybody
I downloaded and installed the 2.0.20 in my 9x turnigy
when the radio is switched on the first warning is the engine not in position, the calibration is done and the gas stick is completely closed. I advise you to press a button.
After this warning comes the second CONTROLLI THR and AIL are selected as if the swich is in position from 0 to 1, then active. He tells me to press a button.
Pressed brings me back to calibrate the sticks for the fifth time.