opentx/opentx #6047: OpenTx & Companion Sim 2.2.2 RC4 - Elevator trim not working

LapinFou, 2018-07-14:
Hi Folks,

I don't understand why, on the attached prog using the elevators trim doesn't move the cursor on the screen.
The trim seems to be taken in account, but the cursor doesn't move. However trims are working fine on Ail and Rud.
The issue can be reproduce on the radio and the simulator.

[Elev_Issue.zip](https://github.com/opentx/opentx/files/2195174/Elev_Issue.zip)

kilrah, 2018-07-14:
Becasue of SF16. When active the  trim adjusts the GVAR instead of trimming.

closed 2018-07-14 by kilrah

LapinFou, 2018-07-14:
Thanks you Kilrah!!! I just discover than my RS slider wasn't calibrate correctly. 
Moreover, to be completely honest, I forgot to move the RS slider on the simulator.