opentx/opentx #6050: increase RSSI data rate

Nihonski, 2018-07-17:
The S.Port Mirror of my X9Dplus provides the RSSI telemetry data about 2 times a second, equal to all other sensor data.
Is there any way to increase the data/refresh rate for RSSI data only?
Could the LUA command "sportTelemetryPush()" support such function?

I implemented an RF closed loop controll, which regulates the RF output from 1mW to 100mW of my transmitter. Higher RSSI data rate would allow me to improve the response time of the loop filter essentially.

3djc, 2018-07-17:
It is comming from the rssi source (module/receiver) . If you need it faster then your source need to send faster

kilrah, 2018-07-17:
S. Port mirror just copies every byte received from the module to the serial port. If you're not getting more, then there is no more.

closed 2018-07-17 by kilrah