opentx/opentx #6051: Companion details

Roitellet, 2018-07-18:
As tested in Opentx 2.2.2RC4:

Some minor handling issues or improvement suggestions:

1) Newly given Curve name does not show in pull down list of curves in OUTPUTS section.
- Newly given curve name is shown only after saving model, closing and opening it again (not ok)
- Newly given Curve name shows immediately in the Edit section of a Channel in MIXER  (ok)

2) Sometimes I have several *.otx files open and their models opened. 
Problem: The header of the model window just provides the model name.
Suggestion: Is it possible to add the *.otx file name at the top of the model window. 
Reason: With the *.otx filenames I know which version I have open (e.g., when copying something from an older backup version to the latest version). 

3) Trying to understand bugs in mixing actions in simulation mode 
Problem: A model is configured and in simulation mode and something does not look right. It is not possible to look up the MIXER section or CURVES while SIMULATION is active. The simulation window needs to be closed to access the code. 
Suggestion: a) When in Simulation mode, allow that the Model window in the background to be activated to come to the foreground. b) allow to switch sections and scrolling in the model window while simulation is active. No editing of any code should be allowed while in simulation mode.
Reason: sometimes switches are set during a simulation (e.g., motor activated) which have to be reset after exiting/entering simulation mode after looking up the code. 

5) Suggestion: copy/paste of special functions from one model to another
Problem: some special functions are always the same. It would be helpful to be able to copy those from one model to another.

Thanks for the excellent work! 
OpenTx is fun to use

mpaperno, 2018-07-18:
1. Confirmed, please submit a individual issue with bug report (or turn this one into it). BTW do not need to save the model, only re-open the editor dialog.
2. Please submit individual feature request issue (I can't find an existing one for this, but please check again).
3. Duplicate of #3056.
4. You can already do that, though currently only one at a time. Related: #4178 (possibly others)

mpaperno, 2018-07-20:
Closed in favor of #6054.

In the future please start individual Issues for unrelated items, it's much easier to track that way.

Thanks!

closed 2018-07-20 by mpaperno