opentx/opentx #6058: 2.2.2 RC5 X7 flashing options sorting

janekx, 2018-07-23:
I have found that in 2.2.2 is diferent flashing sorting then 2.2.1. 
In 2.2.2 when i eleft file from SD hold the enter see flash internal module and then flash external device
in 2.2.1 is external device flash on first line 

I recommend to change like in 2.2.1 to have first external device flash, firstly it is known sorting from 2.2.1 and we this way prevent flashing bad FW to internal module by accidentally pressing enter more time.

3djc, 2018-07-23:
?

[image]

kilrah, 2018-07-23:
He probably has a first bach X7 without bottom connector.

janekx, 2018-07-24:
Yes first ones batch have diferent sorting, any way in newer should be as well  S.Port, External, Internal.

3djc, 2018-07-24:
After discussion with @bsongis , we have made that change, it will be in release version

closed 2018-07-24 by 3djc

janekx, 2018-07-26:
In fact I meet many bad flashed radios even the option flash internal module is on second or third place, maybe will be good to add warning like, "Warning flashing internal module with incorrect firmware can make internal module inoperate." Yes Enter or other key to cancel.